Molodos/ElegooNeptuneThumbnails #4: No image generated / info displayed, but start with high %

ilpantera, 2023-06-02:
First of all: Thank you very much for the great work. Saw the screenshots and it looks great! I have installed it but am facing an issue: Neither the image, nor the info is displayed on my Neptune 3 Pro, however the print starts with 40-50% on the progress bar. As soon as I delete
```
;includeThumbnail
;includeTimeEstimate
;includeFilamentGramsEstimate
;includeLayerHeight
```
From the Start G-code, everything is back to normal. Any idea? Thanks a lot!

### References:
**Start G-code:**
```
;includeThumbnail
;includeTimeEstimate
;includeFilamentGramsEstimate
;includeLayerHeight
G28 ;home
G92 E0 ;Reset Extruder
G1 Z4.0 F3000 ;Move Z Axis up
G92 E0 ;Reset Extruder
G1 X1.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X1.1 Y80.0 Z0.28 F1500.0 E10 ;Draw the first line
G1 X1.4 Y80.0 Z0.28 F5000.0 ;Move to side a little
G1 X1.4 Y20 Z0.28 F1500.0 E20 ;Draw the second line
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
```

**End G-code**
```
G91 ;Relative positioning
G1 E-2 F2700 ;Retract a bit
G1 E-2 Z0.2 F1600 ;Retract and raise Z
G1 X5 Y5 F3000 ;Wipe out
G1 Z10 ;Raise Z more
G90 ;Absolute positionning
G1 X0 Y{machine_depth} ;Present print
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed
M84 X Y E ;Disable all steppers but Z
```

Molodos, 2023-06-02:
Hi,
interesting issue indeed. First of all thanks for all the details. I found out, that they look exactly like mine and I also use the Neptune 3 Pro. Therefore the Problem must be somewhere else.
Can you tell me which Cura version and plugins (click on the Marketplace button and then the settings wheel to view installed plugins) you have installed? I am thinking, that other plugins might mess with mine.
Also, you could send me the created G-code file if you want to. I think that could give some hints on what the issue is.
Best regards

ilpantera, 2023-06-02:
Hey, yes super strange issue. Thanks for checking the G-code.

Cura Version: 5.3.1 - no other plugins installed
Neptune 3 Pro Firmware: 1.1.5.1a 
UI Version V1.4.1

Here is how the screen looks like right after clicking "print": [https://1drv.ms/i/s!AnDUX6Ggc6C3go8eUGqPsvYVEcrM3A?e=3JJguh](url)

Should all be the latest?

Molodos, 2023-06-02:
Actually there is a newer Firmware available:
Mainboard: 1.1.5.1b
UI: V1.4.2
But as yours is pretty new, too, I think that shouldn't be a problem. If it was, then more people would have problems.

Would you be so kind to send me the gcode, that was generated (the one on the sd card you were printing)? I think this might be the only chance to find the problem as everything else seems pretty much like my working setup. If you do so, I could try starting the print on my printer and check if I can reproduce the error.

ilpantera, 2023-06-02:
Thank you! Here is the very simple file as an example showing the above error (with plugin) and working fine (without plugin).

[With Plugin](https://1drv.ms/u/s!AnDUX6Ggc6C3go8k6S9pBbmM6ZFmMw?e=NYabNA)
[Without Plugin](https://1drv.ms/u/s!AnDUX6Ggc6C3go8lAoe6k6zybES5XQ?e=Yy3PRX)

Molodos, 2023-06-02:
Thanks for the files.
I just had a quick look and found out, that the Thumbnail is indeed formatted wrongly. Thats good news I guess. I am currently away and will only be able to try hunt down the bug the day after tomorrow. Until then, I have one more thing that might help me: Could you generate G-code for your Neptune 2 printer that is working with the same object and thumbnail enabled and also send it to me?
Sorry for the many things I need, but I think we are on a good way to hunt down the issue :D

ilpantera, 2023-06-03:
No worries, I'm in no hurry - thanks for taking the time to look into it. 

I think there is a misunderstanding. I don't have a Neptune 2.

Molodos, 2023-06-03:
I am sorry confused you with another one who hat a neptune 2. I will look into everything tomorrow. (Btw: I am also on Win 11)

ChrisCopter, 2023-06-04:
I have the same Issue on the N3P.
There is a attached GCode File with following start code:

;includeThumbnail
;includeFilamentGramsEstimate
;includeLayerHeight
;includeTimeEstimate
G90 ;Absolute positioning
G28 ;Home
;G29 ;Auto bed leveling
M420 S1 Z1;enable correction and compensation height
G92 E0 ;Reset Extruder
G1 Z0.45 F300 ;Move Z Axis up
G1 X1.5 Y20 F5000.0 ;Move to start position
G1 Y120.0 F600.0 E15 ;Draw the first line
G1 X0.5 F1000.0 ;Move to side a little
G1 Y20 F600 E30 ;Draw the second line
G92 E0 ;Reset Extruder
G1 Z40.0 F3000 ;Move Z Axis up
M900 K0 ;Turn off linear advance
M412 S0 ;disable filament runout sensor
M413 S0 ;disable power loss recovery

[Thumbnail.zip](https://github.com/Molodos/ElegooNeptune3Thumbnails/files/11645368/Thumbnail.zip)

Molodos, 2023-06-04:
> I have the same Issue on the N3P. Cura 5.3.0 Neptune 3 Pro Firmware: 1.1.5.1a UI Version V1.4.1
> 
> There is a attached GCode File with following start code:
> 
> ;includeThumbnail ;includeFilamentGramsEstimate ;includeLayerHeight ;includeTimeEstimate G90 ;Absolute positioning G28 ;Home ;G29 ;Auto bed leveling M420 S1 Z1;enable correction and compensation height G92 E0 ;Reset Extruder G1 Z0.45 F300 ;Move Z Axis up G1 X1.5 Y20 F5000.0 ;Move to start position G1 Y120.0 F600.0 E15 ;Draw the first line G1 X0.5 F1000.0 ;Move to side a little G1 Y20 F600 E30 ;Draw the second line G92 E0 ;Reset Extruder G1 Z40.0 F3000 ;Move Z Axis up M900 K0 ;Turn off linear advance M412 S0 ;disable filament runout sensor M413 S0 ;disable power loss recovery
> 
> [Thumbnail.zip](https://github.com/Molodos/ElegooNeptune3Thumbnails/files/11645368/Thumbnail.zip)

I think your issue is different. Are you sure, that your printer is also starting at 50% progress? If I look into your G-code, it does seem like another problem as there are no clues pointing towards this issue. I am happy too help, but if I am right and your issue is a different one, please open a new Issue for that, thank you :D

Molodos, 2023-06-04:
@ilpantera I think I have found the issue. You seem to have used an Ender 3 printer as a base for your printer profile. As my plugin needs to find out, which printer model is used (different Neptune printers use different thumbnail formats), this will lead to errors. I have planned to release a fix, that no thumbnail is generated instead of leading to display errors, but you can fix it for you by using the official Elegoo Neptune 3 Pro printer definition, that comes with Cura, as a base.
Please let me know if that fixes it for you :D

ilpantera, 2023-06-04:
Ahh, thank you makes sense! Where can I find the Elegoo Cura config? This is one that was shared on YouTube for the Neptune 3 Pro as working well (and it did so far).

Molodos, 2023-06-04:
Just open tha Add printer dialog in Cura and select "Non UltiMaker printer" >"Add a non-networked printer" >"ELEGOO" > "ELEGOO NEPTUNE 3 PRO".
I know some time ago you needed to use the Ender 3 as base because there were not Elegoo definitions in Cura. Bu now they were added so you can just use them :)

Molodos, 2023-06-07:
Happy to hear that.
To be honest, I have no idea on what could cause the % error. But if it works without the options for the plugin added to the start G-code, one could compare the two G-code files and might find something. I guess there should be some differences that would give some hints on the error. If you want to, you can also send me those and I will have a look if I find some time :D

Molodos, 2023-06-14:
As there ius no more activity, i will close this issue as fixed :D

closed 2023-06-14 by Molodos