Molodos/ElegooNeptuneThumbnails #5: If you reslice the same body with diff. settings, image (and est. time) won't get updated

silentgecko, 2023-06-03:
I had this issues a couple of days ago.

I "misssliced" (what a word) an object with wrong settings for printing. The estimated print time was 9 mins (in the generated preview image from your plugin).

After it printed i resliced the same object (just switching material profile) and the estimated print time was 20 min (higher detail, more layers).
But the image still shows 9 min as estimate.

Nothing big, but i gues there's a caching issue regarding generating the image

Molodos, 2023-06-03:
Thanks for the info, I will look into it tomorrow. I hope I can reproduce it. If I can't, the G-code of the bugged Thumbnail might help me, but i'll inform you tomorrow, if I need it :)

Molodos, 2023-06-04:
Hi, unfortunately I couldn't reproduce the error. Could you please give me some further information:
- Your Cura Version
- The g-code with the messed up thumbnail
- Info on what the thumbnail displays vs. what it should display
I hope that this information could be enough for me to hunt down the bug :D