Molodos/ElegooNeptuneThumbnails #6: Feature Request: Add Model Height as option

silentgecko, 2023-06-03:
One more idea that i had in mind:

Add the model height as option. For example i look always on the "current layer height" and roughly estimate how far it is in the print.
Because the percentage shows i guess only lines of gcode executed? dunno. But it happened to me quite often that the first 20% took the same time as the last 80%. So the percentage is not correlating to time.

I hope you understand what i mean :D If i can see how many mm my model is (for e.g. 32mm) and see the printer is on layer height 18.2mm i can bestter estimate how far it is in the print :)

But maybe this is not an issue anymore, since i can see the total needed minutes with your plugin 🤔 i digress 😄

Molodos, 2023-06-03:
That was exactly why I added the total estimated Minutes :)
But I think I could accomplish the height. Better one option more as you don't need to use all of them. I will try to add it in the next update :D

Molodos, 2023-06-04:
Just added the feature in the newly released version 2.1.0 :D

closed 2023-06-04 by Molodos