Molodos/ElegooNeptuneThumbnails #10: Thumbnail dont change.

ChrisCopter, 2023-06-14:
I sliced a model and scaled it down, than printed it, everything is shown on the display as intended.
Then i cleared the buildplate load the same model as before and scalid it down to another scale than the first time.
This time the same height and time is shown as the first time, regardless the model is now bigger and takes longer to print.

Molodos, 2023-06-14:
That seems to be a duplicate of #5. Would you be so kind to send me the two g-Code files of before (where everything was fine) and after (when the image stayed the same) to help me hunt down the error? Of course you can also reproduce it with another model and send me those files :D

ChrisCopter, 2023-06-14:
[Treetrunk.zip](https://github.com/Molodos/ElegooNeptuneThumbnails/files/11750568/Treetrunk.zip)

ChrisCopter, 2023-06-14:
[Treetrunk.zip](https://github.com/Molodos/ElegooNeptuneThumbnails/files/11750624/Treetrunk.zip)

Ive loaded another model after clearing the buildplate and the trunk is shown as before on the printer.
Thank you for your work, i love the addon.

Molodos, 2023-06-14:
Okay, so I just tried finding something out, but unfortunately had no success yet. The following things might help me:
- What Cura Version are you using?
- Do you have any other plugins installed and if yes, which?
- If you can reproduce the error once more and then instantly close Cura and send me the cura.log file, that could also help me massively. For me (Windows) the log file is located under "C:\Users\<username>\AppData\Roaming\cura\5.3\cura.log"
Thanks for your feedback and support, I hope we can fix that bug :D

ChrisCopter, 2023-06-14:
[Treetrunk.zip](https://github.com/Molodos/ElegooNeptuneThumbnails/files/11751010/Treetrunk.zip)

There you go.
Iam on Cura 5.3
Addons installed:
Auto Towers Generator
Calibration Shapes
Linear Advance Settings
Settings Guide

Molodos, 2023-06-15:
Okay, I have tried to collect some data from the files you shared, but unfortunately I do not quite know which file is which. It also seems like wen reproducing, you only did one slice instead of two as there was only one new gcode file in the zip.

Also you might want to try updating to Cura version 5.3.1 and the newest plugin version, it seems like you are using 2.1.0 atm. If the bug then still persists, I am happy to analyze log and the two gcode fiels again if I know which is which :D

closed 2023-06-22 by Molodos

silentgecko, 2023-08-05:
Heya, sorry for not responding.

I'm quite sure I know where the problem is now.
It's not in the gcode or in your plugin, it's in the Neptune 3 firmware itself.

When I print one of the models sliced with your add-on and it finishes successfully, and if I then immediately print another model with a complete other picture and data, then the image is not "refreshed" or reloaded inside the display of my Neptune.
When I stop a print or when I completely, reboot the printer, them the correct picture is shown - always.

So to reproduce just slice for e.g. 2 diff. Models and put them in one folder. After successful printing the first, start the second. It will still show the image of the first model.

Hope this helps 😅
Kr

Molodos, 2023-08-05:
Hi,
thanks for the info. I can say, that I did never experience such problems. You might wanna look at your firmware version of the printer and check if there is an update available. If there is one, there are good chances that it will fix the thumbnail problem :D