Molodos/ElegooNeptuneThumbnails #12: Broken thumb on large detailed object files.

Gareee, 2023-07-29:
I sliced a very detailed object and saved the slice. The thumb would not display on the neptune 3 max.
I attempted a re slice with the same issue,  andvtried a resave just in case the save was corrupt. 
The resave didn't work, but the file size was significantly larger. I kept reading with only numbers filename changes, and each save was larger, but none would display the thumb size.
I suspect either the thumb has a size restriction, or the thumbnail plug in doesn't work with highly detailed objects. (It takes 5-10 minutes to slice.)

Molodos, 2023-07-29:
Hi,
interesting error. Do you haver a Cura Project file for me? Also a G-code File might help me to analyze the error, too.

Gareee, 2023-07-29:
I can send the obj and slices. Every time I saved, the file size increased. It looked like the thumb image codes were increasing in size. This email good for the files?Sincerely,Gary Tabar Jr.

Molodos, 2023-07-30:
I am not sure, but just try. Otherwise you can just drag and drop them when answering to this issue on GitHub :D

Gareee, 2023-07-30:
Tried dragging and dropping and even two sliced files were too big zipped for attachement. I’m including a few sliced gcodes and the original obj file attached in this reply.

EDIT: Too big for email attachments. Adding a google drive link. Please Lmk once you have it so I can delete it.

https://drive.google.com/file/d/1Q9zEkTEE3KmQgpmnvMZ8s3y5qWW5eHwY/view?usp=drive_link

From: Michael Weichenrieder
Sent: Sunday, July 30, 2023 2:47 PM
To: Molodos/ElegooNeptuneThumbnails
Cc: Gareee; Author
Subject: Re: [Molodos/ElegooNeptuneThumbnails] Broken thumb on large detailedobject files. (Issue #12)

I am not sure, but just try. Otherwise you can just drag and drop them when answering to this issue on GitHub :D

Molodos, 2023-07-30:
Hi, just sent you an access request on google drive. Let me know when you accepted so I can let you know when I downloaded :D

Gareee, 2023-07-30:
I think it's shared now...been a while since I really used Google drive. Sincerely,Gary Tabar Jr.

Molodos, 2023-07-30:
Okay, I downloaded it. I will get back to you as soon as I found time to analyze it

Gareee, 2023-07-30:
Cool. Might be a memory thing...cura was taking an awfully long time slicing it, and I have a decent pc I 3dcsculpt on and play games on, with 32 gb ram.Sincerely,Gary Tabar Jr.

Molodos, 2023-08-07:
Okay, so current state is that I found out, that every slice seems to add a thumbnail to the G-code so that in the end there are multiple thumbnails in the G-code. I don't know how to fix it yet and the no-display error could be part of it, but I just wanted to let you know, that I am working on it

Gareee, 2023-08-07:
Cool. Elegoo told me I should report it to you. We just got ours a month ago so I just assumed it was on the latest firmware. Sincerely,Gary Tabar Jr.

Molodos, 2023-08-07:
Oh interesting, so you contacted the Elegoo support and they redirected you to me?
I just published a new release here on GitHub (https://github.com/Molodos/ElegooNeptuneThumbnails/releases/tag/2.2.0) it fixes the file size increase bug. Feel free to install it and check if the thumbnail now also displays correctly.
Best regards

Molodos, 2023-08-07:
Thanks for the detailed description on what happened. I always love to hear such things. I am happy to hear your feedback after you tested the new version :)

Molodos, 2023-08-10:
Hi, just wanted to know whether you have testes the newest version and if it fixes the bug so I can close the issue

Gareee, 2023-08-13:
> Okay. I will close the issue for now, feel free to still give me feedback if the bug was fixed

Sorry...printer issues and wife on vacation. ;) I'll try to look at it tomorrow.Sincerely,Gary Tabar Jr.

Molodos, 2023-08-13:
Oh okay, no problem :)

Gareee, 2023-08-30:
I can confirm the fix works across multiple files now. I ran into another one, updated, and then saved the older ones that wouldn’t work along with the new file, and all showed the thumb properly on the Neptune 3 max. Yay!

(Sorry this took so long to check, but life finds a way…lol!)

From: Michael Weichenrieder
Sent: Sunday, August 13, 2023 6:09 PM
To: Molodos/ElegooNeptuneThumbnails
Cc: Gareee; Author
Subject: Re: [Molodos/ElegooNeptuneThumbnails] Broken thumb on large detailedobject files. (Issue #12)

Oh okay, no problem :)