Molodos/ElegooNeptuneThumbnails #13: Plugin interaction with ultimaker printers causes an error when saving a 3mf

HellAholic, 2023-08-14:
**General information (please complete the following information):**
 - Printer: [Ultimaker S5]
 - Cura Version: [5.4.0+]
 - Plugin Version: 2.1.3

**Describe the bug:**

If plugin is installed and you slice with Ultimaker S5 printer selected, when saving the slice on disk, it gives an error 'time'

Saves normally when slicing with Elgoo Neptune printer.

**Steps to reproduce:**
1. Load any stl
2. Select an Ultimaker printer
3. Slice the model
4. Save on disk
5. Error ['time']

**Additional information:**
Add additional information (optional)

Molodos, 2023-08-14:
Hi, thanks for the report. I have successfully reproduced the bug and am currently working on a major update, so with version 3.0.0, the bug will be fixed. I'm not sure if I manage to release the new version before my vacation so expect it in the next few days or in about 2 Weeks :)

Molodos, 2023-08-15:
Version 3.0.0 is out, bug should be fixed. If it isn't, don't hesitate to re-open the issue :)