Molodos/ElegooNeptuneThumbnails #14: Neptune 4 Pro Thumbnail dont show

BremerButjer, 2023-08-15:
**General information (please complete the following information):**
 - Printer: Neptune 4 Pro
 - Printer Firmeware Version: 1.8
 - Cura Version: 5.4.0
 - Plugin Version: 2.1.3 shown in Cura, but 2.2 manual installed

**Describe the bug:**
Estimated time is given as 0 min. Percentage starts with any value.
The preview image is displayed completely black without content.

Molodos, 2023-08-15:
Hi, earlier today, I just released plugin version 3.0.0 with many major changes and fixes. Pleas check it out and see if it will fix the bug you are having. If it doesn't, I am happy to help :)

Molodos, 2023-08-15:
(PS: I just saw, that you are already using a Neptune 4 Pro, which unfortunately is not already officially supported by my plugin. You could be fine by selecting the Neptune 3 in the plugin menu instead, but I cannot guarantee. But I can let you know that I am already in contact with Elegoo in order to officially support the Neptune 4 (Pro) as soon as possible)

Molodos, 2023-08-16:
Quick update: The Neptune 4 will be added in about 1-2 weeks :)

tripletto, 2023-08-23:
Neptune 4 here: Your plugin version 3.0.0 works like a charm with this printer and Cura 5.4.0. Thank you!

Molodos, 2023-08-23:
> Neptune 4 here: Your plugin version 3.0.0 works like a charm with this printer and Cura 5.4.0. Thank you!

Happy to hear that. I'm now officially sponsored by Elegoo and got hands on a Neptune 4, so official support is also not far away

Molodos, 2023-08-26:
Good news: Neptune 4 support is out :)

closed 2023-08-26 by Molodos