Molodos/ElegooNeptuneThumbnails #15: Version 3.0.0 makes black square on the display in stead of model's picture.

UvarovAndrei, 2023-08-16:
**General information (please complete the following information):**
 - Printer:Elegoo Neptune 3 Pro
 - Printer Firmeware Version: 1.1.5.1a
 - Printer UI version V1.4.1
 - Cura Version: 5.4.0
 - Plugin Version: 3.0.0

**Describe the bug:**
The plugin 3.0.0 creates just a black square on the display of the printer.

**Steps to reproduce:**
1. Load the 230812_Hook.stl model in Cura
2. Slice it
3. Save the 230812_Hook.gcode file to external TF memory card
4. Insert the card to printer and start to print.

**Additional information:**
If just to delete the first (;start_thumbnail) and the last (;end_thumbnail) rows in .gcode file, it start doing well.
The picture of the model appears and all the chosen data as well.
For example file Hook_1 shows black square, and the file Hook_2 shows all right.

[230812_Hook.zip](https://github.com/Molodos/ElegooNeptuneThumbnails/files/12358369/230812_Hook.zip)

Molodos, 2023-08-16:
Hi, thank you for the report and also for already finding the cause in the gcode. Unfortunately I am on vacation right now, so I will be able to fix the bug in about a week. If you already need thumbnails earlier, please consider downgrading the plugin version to 2.2.0 until then :)

Molodos, 2023-08-16:
Good news: Couldn't wait and found a quick fix. Version 3.0.1 is out. I would be happy if you could give me feedback, if the bug was fixed as I couldn't try it on a printer. But I think your description was good enough to fix it. Thanks again, mate :)

UvarovAndrei, 2023-08-16:
I was watching football, but couldn't wait as well.
So i tried. 
It is okay now, what a pleasure!
Thank you!

closed 2023-08-16 by UvarovAndrei

Molodos, 2023-08-16:
Happy to hear that :)