Molodos/ElegooNeptuneThumbnails #16: Neptune 4 Pro Support

iosdoctur, 2023-08-25:
**Describe the feature you would like to have added: Neptune 4 Pro support
A clear and concise description of what you want to be added

**Describe why the above feature should be added:
Currently unable to view thumbnail on Neptune 4 Pro when using Cura
A clear and concise description of why the feature is relevant

**Additional information:**
Add additional information (optional)
Please if you can add it’ll be awesome!

Molodos, 2023-08-25:
Good news: I am currently working on it. Expect the update to be available in less than a week :)

iosdoctur, 2023-08-25:
Thank you for the quick response and that is great news, I’m definitely looking forward to it!

Thank you,
Haris

Molodos, 2023-08-26:
Just published the update that adds neptune 4 support :)

closed 2023-08-26 by Molodos

iosdoctur, 2023-08-26:
Michael!
Thank you so much man, I really appreciate the urgency on this, can’t tell you enough how grateful I am.I I recently bought a Neptune 4 pro and was not aware of how new it is in the market with little support. I just tested the plugin and it works great, love how you’ve given users the ability to customize what they want displayed in each corner. I also attached a preview of my printer working with the plugin.

Thank you,
Haris

Molodos, 2023-08-26:
Thanks mate, happy to hear that :)