Molodos/ElegooNeptuneThumbnails #19: Thumbnails do not appear on the ELEGOO Neptune 2.

ImBadAtNaming, 2023-09-17:
**General information (please complete the following information):**
 - Printer: ELEGOO Neptune 2
 - Printer Firmeware Version: V2.0.1
 - Cura Version: 5.1.1
 - Plugin Version: 3.2.1
 - Operating System: Windows 11

**Describe the bug:**
When using the plugin to generate thumbnails on for the Neptune 2, the generated thumbnails do not appear.

**Steps to reproduce:**
1. Load any model and slice.
3. Save file to sd card.
4. Insert sd card into printer. No thumbnail appears on file selection or print screen.

**Additional information:**
I believe the issue is caused by the thumbnail generated by the plugin being formatted incorrectly. 

To make the plugin match ELEGOO's thumbnails:
1. ;gimage should become ;;gimage
2. Swap the positions of both images in the gcode.
3. ;simage should be 409 columns wide for each row in the gcode file, and ;;gimage should be 809 columns wide per row (first line is 810 columns).

I discovered this by comparing the thumbnails generated by the official ELEGOO Cura and the ones from the plugin.
I have linked pastes of the thumbnails below so you can compare the two:

Plugin: pastebin.com/RUbpXSBG
ELEGOO Cura: pastebin.com/mu3ZSwpb

Molodos, 2023-09-17:
Hi,
first of all thank you so much for analyzing the issue that deep, you saved me some work there. As far as I know, you nailed it and I already published a fix version (Version 3.2.2). As I do not have a Neptune 2 to check everything, I am not able to test it. I would love to hear your feedback, if the bug is fixed and will leave the issue open until then.
Best regards

ImBadAtNaming, 2023-09-18:
Hello,

Thank you for responding so quickly!
Thumbnails now appear on the printer, but there is still an issue.
The thumbnails image data seems to be incorrectly formatted, causing the printer to draw them incorrectly.
I have attached photos of the problem.

[image]
[image]

Molodos, 2023-09-18:
Hi,
thank you for the fast and informative feedback. I have looked into everything again and found another small bug and fixed it with release 3.2.3. I would be happy if you could try it once again :)

ImBadAtNaming, 2023-09-19:
Hello,

I am still getting the same result.
I have noticed that the data for the top image (;simage) in the gcode is 240 columns too wide.
Can you try fixing that? I believe it may be a primary factor in the cause of the bug.

Thank you.

Molodos, 2023-09-19:
Hi,
thanks for noting that again (I had already checked it but couldn't find to fix it then). Now, I finally found the issue causing the lines to be too much and too long (because of a too high resolution for the Neptune 2). Once again, I just published a new release but this time I am pretty sure, that the bug will finally be fixed. Happy to hear your feedback.
Best regards

ImBadAtNaming, 2023-09-19:
Hi,

I can confirm that the bug is fixed, and thumbnails are now properly displayed.

Thank you for fixing it!

Molodos, 2023-09-20:
Happy to heat that. Thanks for your help to fix it :)

closed 2023-09-20 by Molodos