Molodos/ElegooNeptuneThumbnails #20: Not showing up under extensions tab

GSMaxxx, 2023-09-19:
**Question:**
The extension isn't showing up under the extensions tab,

**Additional information:**
I've tried restarting,  it just refuses to show up, any ideas? downloaded from the marketplace

Molodos, 2023-09-19:
The Marketplace Version is outdated because Ultimaker takes Months to verify an Update (thats ridiculous). Therefore the GUI is only in the GitHub version for now, thry installing it that way :)

closed 2023-09-19 by Molodos

GSMaxxx, 2023-09-19:
i did try that, doesn't seem to work either :'( i dragged the base.curaprofile into cura and restarted

GSMaxxx, 2023-09-20:
That's totally my bad, i realized the name was different but i always forget the releases tab. Still, i downloaded from the link and dragged it, still didnt work :( i have no idea what im doing wrong now