Molodos/ElegooNeptuneThumbnails #21: Conflict with other printers?

GregValiant, 2023-10-01:
**General information (please complete the following information):**
 - Printer: UM3
 - Printer Firmeware Version: N/A
 - Plugin Version: UM Cura Marketplace
 - Operating System:  Windows 11

**Describe the bug:**
Bug report on the UM Cura site [#16862](https://github.com/Ultimaker/Cura/issues/16862) on the UM Cura site seems to indicate a problem with a "time" error when this post processor is enabled and the OP makes his UM3 the active printer.

**Steps to reproduce:**
I'm guessing here...
Install the post processor.
Switch printers from the Elegoo to a UM3 (or other printer that doesn't support thumbnails??)
Slicing results in a - 'time' error

Molodos, 2023-10-01:
That's a duplicate of #13 , which was fixed in Version 3.0.0. Please consider updating the plugin :)
Current plugin versions are 3.0.1 on Marketplace and 3.2.4 on GitHub (unfortunately Ultimaker will take more than a Month for review, which makes updating the plugin kind of a pain for me)

closed 2023-10-01 by Molodos

GregValiant, 2023-10-01:
In my version of 5.4.0 the current MarketPlace version of the plugin is 2.1.3.  That appears to be well out of date.  I'll leave word of that in the bug report on UM Git.

Molodos, 2023-10-02:
Thats interesting. Maybe it takes even longer for users to see the update. For reference: I sent version 3.0.1 out for approval to Ultimaker about 6 weeks ago. Yesterday I checked and it was finally approved, so I published version 3.0.1 and immediately send out 3.2.4 for approval since there were many updates within that 6 weeks of waiting. So maybe 3.0.1 is now visible to you after some more waiting, but all in all I would love the review process to be a bit faster.

GregValiant, 2023-10-02:
Today I see 3.0.1in the MarketPlace so maybe somebody saw the post on the Cura Git this morning.
I have no idea how they are organized at UM, but it isn't a big team and they have Cura, the pull requests, the website, the forum, the WIKI, UM firmware, etc., that need maintenance.  It doesn't surprise me that it takes a while but some "feature" requests have been around for 6 or 7 years.  I suppose it boils down to how they prioritize internally.