Molodos/ElegooNeptuneThumbnails #22: Cura SDK upgrade, but no contact information

rburema, 2023-10-11:
**Question:**
Hi! Cura dev here. You've not listed any contact information, which we use to inform plugin-contributors of upcoming Cura versions -- we're releasing 5.5 soon-ish (in fact, [the beta](https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1) is already out). None of the changes should be breaking with the 8.x.x SDK (minor upgrade from 8.4.0 to 8.5.0), but it might be a good idea to test your plugin(s) with the upcoming version anyway via the beta.

**Additional information:**
It's ok to not have the contact information there if you don't want to, but we're probably not going to make the effort each time to submit an issue on github either.

The original mail follows:

> Dear Cura-plugin contributor,
> 
> You've been sent this mail because this mail-address was given to us as the correspondence address on at least one currently active (or submitted, pending approval) UltiMaker Cura plugin.
> 
> This mail is to inform you that, for the upcoming Cura version (5.4.x), we're bumping the SDK/API version to '8.5.0' (was '8.4.0').
> 
> (The Cura 5.5-BETA will be out very shortly.)
> 
> Since this is a minor increase, there shouldn't be any breaking changes. This also marks the first time the nascent 'Engine Plugins' feature is unveiled. You may want to update, depending on whether the added classes or updated function signatures (added optional parameters or type-checking) are of use to your plugin.
> 
> Changes accessible for SDK-usage in Uranium and Cura made after 5.4.x are (as always) added to the wiki, here: https://github.com/Ultimaker/Cura/wiki/SDK-Versions
> 
> kind regards,

Molodos, 2023-10-11:
Hi,
thanks for notifying me, I will test my plugin against the prelease as soon as possible.
Also, could you tell me where exactly I need to add my contact info and what information is needed in order to receive that notifications in the future? Is it a email in the plugin.json?
Thanks and best regards!

rburema, 2023-10-27:
> Also, could you tell me where exactly I need to add my contact info and what information is needed in order to receive that notifications in the future? Is it a email in the plugin.json?

Sorry for the late response, we use the email given to us when registering in the community portal.

Molodos, 2023-10-27:
Okay, good to know. I just opened the community portal (until now I only have used the contributor portal) and had to give some more info, but not an email. Maybe my community account was now created from my Ultimaker account and it could now work. Do you have some way too look up, if the contact email is now accessible for the future?
Thank you :)

rburema, 2023-10-27:
Hm, then you might be right in that it's the field from the plugin instead. Here's what I see when I'm administrating the plugin portal, for your plugin (next to the envelope icon you'd expect an email address).

[image]

Molodos, 2023-10-27:
Oh, I think I know the problem. The first time I created my Ultimaker account with the username "Molodos" I did never get an activation link via email (looks like the system has got problems when sending to "...@t-online.de"). Therefore I created another account "RealMolodos" with another email and it worked. But as I want my nickname, I used "Molodos" in the plugin.json, which probable led to connecting the plugin to the wrong account which has no email, because there never was a confirmation mail. Do you think that is plausible?
And do you know if there is any way of deleting my broken account "Molodos" and then updating the username of my "RealMolodos" account?
Thanks :)

Molodos, 2023-10-27:
Just found the setting in the Ultimaker Contributor portal. Feeling stupid, that it took me so long to find it. If you're curious, I am attaching an image.
[image]
Also just to be safe, could you check once again? Thank you :)

closed 2023-10-27 by Molodos

rburema, 2023-10-27:
I can conform your email is now also visible to us in the portal. Thanks!