Molodos/ElegooNeptuneThumbnails #24: dynlib/dll error: Cura 5.5 with neptune 4 pro

vitofasano, 2023-10-21:
**General information:**
 - Printer: ELEGOO Neptune 4 Pro
 - Printer Firmware Version: [found in printer menu]
 - Cura Version: [5.5.0 beta]
 - Plugin Version: [3.4]
 - Operating System: [MacOS]

If I flag only "enable Klipper thumbnails" this plugin works, but I flag also "enable thumbnails" when I try to save (or send to klipper) g-code following error is shown:

Failed to load dynlib/dll '/Users/vito/Library/Application Support/cura/5.5/plugins/ElegooNeptune3Thumbnails/ElegooNeptune3Thumbnails/tools/../libs/libColPic.dylib'. Most likely this dynlib/dll was not found when the application was frozen.

Molodos, 2023-10-21:
Did you check the path of the lib on your system to see it the lib exists?

vitofasano, 2023-10-22:
Here:

vito@MacBook-Air-di-Vito ElegooNeptune3Thumbnails % pwd
/Users/vito/Library/Application Support/cura/5.5/plugins/ElegooNeptune3Thumbnails/ElegooNeptune3Thumbnails

vito@MacBook-Air-di-Vito ElegooNeptune3Thumbnails % cd tools

vito@MacBook-Air-di-Vito tools % ls -l
total 128
-rw-r--r--  1 vito  staff    314 21 Ott 18:30 __init__.py
-rw-r--r--  1 vito  staff   8412 21 Ott 18:30 gui.py
-rw-r--r--  1 vito  staff  13738 21 Ott 18:30 gui.qml
-rw-r--r--  1 vito  staff   9561 21 Ott 18:30 settings.py
-rw-r--r--  1 vito  staff   1507 21 Ott 18:30 statistics_sender.py
-rw-r--r--  1 vito  staff  12829 21 Ott 18:30 thumbnail_generator.py

vito@MacBook-Air-di-Vito tools % cd ..

vito@MacBook-Air-di-Vito ElegooNeptune3Thumbnails % cd libs

vito@MacBook-Air-di-Vito libs % ls -l
total 1072
-rw-r--r--  1 vito  staff  518118 21 Ott 18:30 ColPic_X64.dll
-rw-r--r--  1 vito  staff   12756 21 Ott 18:30 libColPic.dylib
-rw-r--r--  1 vito  staff   12136 21 Ott 18:30 libColPic.so

vito@MacBook-Air-di-Vito libs % cd ..

vito@MacBook-Air-di-Vito ElegooNeptune3Thumbnails % ls -l
total 136
-rw-r--r--  1 vito  staff  35181 21 Ott 18:30 LICENSE
-rw-r--r--  1 vito  staff   9127 21 Ott 18:30 README.md
-rw-r--r--  1 vito  staff    431 21 Ott 18:30 __init__.py
-rw-r--r--  1 vito  staff   2244 21 Ott 18:30 changelog.txt
-rw-r--r--  1 vito  staff   7263 21 Ott 18:30 elegoo_neptune_thumbnails.py
drwxr-xr-x  8 vito  staff    256 21 Ott 18:30 img
drwxr-xr-x  5 vito  staff    160 21 Ott 18:30 libs
-rw-r--r--  1 vito  staff    566 21 Ott 18:30 plugin.json
drwxr-xr-x  8 vito  staff    256 21 Ott 18:30 tools

Molodos, 2023-10-22:
Okay, so the library seems to be theren. Maybe it is a similar Problem like in https://github.com/Molodos/ElegooNeptuneThumbnails-Prusa/issues/5

vitofasano, 2023-10-22:
So there can't be any changes?

Molodos, 2023-10-22:
What you can do is trying out if your thumbnails will work with Elegoo Cura. It they do, there is a chance. Else not, unfortunately. Feel free to test that and let me know the results if you want to :)

vitofasano, 2023-10-22:
I generated the g-code file with ElegooCura and loaded it both via USB stick and via network klipper, but in both modes the thumbnail was not displayed on the printer's external display.

So if I understand correctly you will not be able to have the thumbnail on the external display of the printer if I use the MacOS version. Right?

Molodos, 2023-10-22:
Yes. Unfortunately I am dependent of the official Elegoo Cura binaries and as there are no binaries for the MacOS M series Chips, I am not able to provide support for those.

Molodos, 2023-10-26:
Okay, just saw, that this seems to be a duplicate of #26. I will do some more tests over in the new issue, feel free to join :D

closed 2023-10-26 by Molodos