Molodos/ElegooNeptuneThumbnails #25: Open Browser did not work

Cascadier1, 2023-10-26:
**General information (please complete the following information):**
 - Printer:  ELEGOO Neptune 4 Pro
 - 1.1.2.03
 - Cura Version: 5.5 beta
 - Plugin Version: dont know
 - Operating System: Win11 Pro

**Describe the bug:**
after sendig file to the Printer, a windows open asking if i want to open Browser did not work (nice to have)

**Steps to reproduce:**
1. Load '...'
2. Slice '...'
3. Save '...'
4. Error

Molodos, 2023-10-26:
Hi, could you please elaborate further on why you think, that this has something to do with my plugin? Because my plugin does not do things like sending files to the printer.

Cascadier1, 2023-10-26:
Hallo Michael,
Das hab ich wahrscheinlich falsch ausgedrückt. Das Plugin läuft super. Das
einzige, was bei mir nicht geht ist: nachdem der Druck erfolgreich an den
Drucker gesendet wurde, geht ein Fenster auf das aus deinem Plugin kommt,
in dem man auswählen kann, ob sich die Webseite dazu öffnet. Und nur dort
geht was nicht bei mir.

Spende kommt!

Bernd

2023 um 19:32:

Cascadier1, 2023-10-26:
Oh, no…
Jetzt verstehe ich erst… Ich hatte das falsche Plugin in Verdacht.  Sorry
🙄

2023 um 19:32: