Molodos/ElegooNeptuneThumbnails #26: ElegooNeptune3Thumbnails Plugin not working anymore under Cura 5.5.0

wesiho, 2023-10-26:
**General information (please complete the following information):**
 - Printer: ELEGOO Neptune 4 Pro
 - Printer Firmeware Version: V1.1.2.30
 - Cura Version: 5.5.0
 - Plugin Version: 3.4.0
 - Operating System: Mac OS 14.0

**Describe the bug:**
After the Cura Update from Version 5.4 to new Version 5.5 the Plugin is not longer working.

The Error Message is:
Failed to load dynlib/dll '/Users/admin/Library/Application Support/cura/5.5/plugins/ElegooNeptune3Thumbnails/ElegooNeptune3Thumbnails/tools/../libs/libColPic.dylib'. Most likely this dynlib/dll was not found when the application was frozen.

Deinstallation and reinstallation does not help-

Molodos, 2023-10-26:
I have already had similar issues. Unfortunately I do not have a Mac and therefore cannot test. Nevertheless, I just have released version 3.4.1 with some changes. Maybe they help, maybe not. Feel free to check it out and give me feedback :)

wesiho, 2023-10-26:
Thank you for your Information. I just testet the new Version 3.4.1 now with same effect and same Error Message. Hope you will find the Problem. Regards, wesiho

Molodos, 2023-10-26:
[ElegooNeptuneThumbnails.zip](https://github.com/Molodos/ElegooNeptuneThumbnails/files/13183486/ElegooNeptuneThumbnails.zip)
I just tried something else. Would be happy if you could try that version, too. The installation package is in the zip file :)

vitofasano, 2023-10-27:
thanks, I'll try it as soon possible

wesiho, 2023-10-27:
This newest version can not be installed in Cura 5.5. on my Mac. Here is the Error Message:
Das Plug-in ElegooNeptune3Thumbnails konnte nicht geladen werden. Die Neuinstallation des Plug-ins könnte das Problem lösen.
I tried it 3 times. Always the same Message.

wesiho, 2023-10-27:
In between I tested the yesterday Version 3.2.4 on Cura 5.4. It works perfect there. The today Version also can not be installed on Cura 5.4 I hope this helps.

Molodos, 2023-10-27:
Thats interesting. So the problem seems to be the combination of Cura 5.5 and using a Mac. Could you by any chance hand me a log file (cura.log) from after the failed installation? Maybe it could help me working out the issue.

wesseli01, 2023-10-27:
Hier die log

[cura.log](https://github.com/Molodos/ElegooNeptuneThumbnails/files/13189152/cura.log)

wesiho, 2023-10-27:
And here is my cura.log
Installierte Version: 3.4.1
Cura Version: 5.5 ARM-Version (M1-Chip) 
[cura.log](https://github.com/Molodos/ElegooNeptuneThumbnails/files/13189859/cura.log)

Molodos, 2023-10-27:
> ElegooNeptuneThumbnails.zip

Thank you very much! Could maybe one of you also check the plugin version 3.4.2 I posted on the linked message and send me a cura.log of that, too. That could be very helpful.

Thank you :)

wesseli01, 2023-10-27:
@Molodos 
Bekomme leider nach der Installation + Neustart der Version 3.4.2 mit Cura 5.5 folgende Fehlermeldung:

[image]

Molodos, 2023-10-27:
Thank you very much. With all that info, I will try finding a solution over the next days, I will contact you here, if I need further assistance

wesseli01, 2023-10-27:
No Thank you for your support. Let me know if you need more information.

Kind regards,
wesseli

wesseli01, 2023-10-27:
Thank you for your support. Let me know if you need more information.

Kind regards,
wesseli

wesiho, 2023-10-27:
With Version 3.4.2 I've got exactly the same Error Message as wesseli01. Hope you will find a Solution for that nice Plugin.

Kind Regards, wesiho

wesiho, 2023-11-03:
Hi. The Problem only occurs with the UltiMaker-Cura-5.5.0-macos-ARM64 version. 
With the Version UltiMaker-Cura-5.5.0-macos-X64 Thumbnail Plugin works perfect. This Version I tested with an M1 Processor on a Mac Mini.

Kind Regards wesiho

smithmatthews, 2023-11-04:
I had this issue as well, after upgrading from Cura 5.4.0 to 5.5.0, Windows 11 23H2 (build 22635.2700), and plugin version 3.2.4 from the marketplace install.  I uninstalled the marketplace version and installed the one in your zip file a few posts up and I can confirm that thumbnails are now working.  Reference attached image.  Left gcode is before performing the steps and preview complains there is no embedded image.  Right gcode is after installed the version from your zip file.

[image]

Molodos, 2023-11-06:
Quick update:
I am currently reverse engineering the Elegoo binaries with the goal of re-writing them in python to get rid of those dependencies that cause errors. Could take some more time, but just wanted to keep you updated :)

demq, 2023-11-10:
> Quick update: I am currently reverse engineering the Elegoo binaries with the goal of re-writing them in python to get rid of those dependencies that cause errors. Could take some more time, but just wanted to keep you updated :)

Hello, 
 Just want to chime in here. I am also running Cura 5.5 on ARM, and keep getting the error when attempting to save the file with the preview enabled:
 "Failed to load dynlib/dll '/Users/yzyz/Library/Application Support/cura/5.5/plugins/ElegooNeptune3Thumbnails/ElegooNeptune3Thumbnails/libs/libColPic.dylib'. Most likely this dynlib/dll was not found when the application was frozen."

I could indeed find libColPic.dylib in the path, and some posts suggest this issue might be caused by ctypes? 

 I am happy to try and create the package on my machine if you could help me figure out some of the dependencies. I had to install https://github.com/Ultimaker/pyArcus as one of the dependencies, but then was getting an error:
-- 
/UM/Backend/SignalSocket.py", line 10, in <module>
    class SignalSocket(Arcus.Socket):
AttributeError: module 'pyArcus' has no attribute 'Socket'
--

Looked like UM requires pyArcus, but there was a mismatch between the versions in their masters....

ruiaraujo012, 2023-11-26:
Same issue, any update?

Molodos, 2023-11-26:
The current state is, that I am in contact with Elegoo and they are saying, that they will work on a documentation for their thumbnail encoding format. As soon as that is out, I will be able to rewrite the libraries, that are causing the errors, in python to fix those kind of issues once and for all.
Unfortunately I do not know how long it is going to take Elegoo to release that documentation. So if anyone wants to speed things up and is good at reverse engineering libraries, feel free to find out how the encoding works and share it here. I appended the library file :)
[libColPic.zip](https://github.com/Molodos/ElegooNeptuneThumbnails/files/13468299/libColPic.zip)

ruiaraujo012, 2023-11-26:
Thank you for your feedback, I'll try to help :)

AshTemp, 2023-11-27:
I suspect there's just something wonky with Cura 5.5, I am able to update the Neptune 4 Max gcode in 5.4 but if I try in 5.5 it just straight up crashes so while not definite proof there's something wrong certainly seems an odd coincidence that trying to add a Neptune 4 is crashing it.

Molodos, 2023-11-27:
I agree, that this is definitely a Problem caused by Cura 5.5. But as the loading of the libraries seems to be the problem, I am also sure, that my above described plan would fix it

sskeledzija, 2023-12-12:
I am not sure which Mac OS versions do you guys use, but could it be that libColPic.dylib does not work the same on all MAC OS versions? Dynamic linker may differently interpret the .dylib file due to different OS version, same file could act differently on different OSs...  I have latest (Sonoma 14) and it also does not work for me, the same error is displayed as you have already described.  Not sure about the dynamic linker details, and how different could it be, but I just wanted to mention if it brings any help, maybe the issue is only in this lib file format..

ruiaraujo012, 2023-12-12:
I think it's not the macOS version but the architecture of the chip, I have the M1 mac.

Molodos, 2023-12-22:
I am happy to announce version 4.0.0 of my plugin, which reworks the core completely in order to hopefully once and for all end the above described bugs from happening.
To everyone here, who had those problems, could you please try updating and give me feedback if it fixed the bug for you? :D

Molodos, 2023-12-22:
> Thank you for your work.
> 
> Will this work to the prusa version as well?

Yes, as soon as I have confirmation, that it works on the Cura version, I will also update the Prusa version

ruiaraujo012, 2023-12-22:
Nice, many thanks, I've switched to Orca Slicer, and it will help to have thumbnails there :)

ruiaraujo012, 2023-12-22:
Hum, ok, it works on Cura, there's no error on generate G-code, but no thumbnail appears on my printer (N4P). I'm using M1 Mac though.

Molodos, 2023-12-22:
I have heared, that there are also some problems depending on the printer firmware. What firmware do you have installed and could you maybe share the gcode, so I can also test in on my N4P it it displays there?

ruiaraujo012, 2023-12-22:
Sure, I'm using:
`Firmware: 1.1.2.41` and `UI: 1.2.11`

Molodos, 2023-12-22:
Thank you! So your firmware is indeed a newer one than I use, that could be it. I will try out your gcode as soon as my current print is finished :D
Meanwhile if you want to, you could try if thumbnails are shown when using the Elegoo Cura slicer. If they show, it is most likely a bug with my plugin, else I guess it is is a bug with the newest printer firmware.

ruiaraujo012, 2023-12-22:
It doesn't work with elegoo cura as well, I thought it could be because of the M1 chip but it could be the firmware.

I'll use a windows pc that I have here at home and try if it works, if so, it could be M1 chip related. I'll post the updates here.

Molodos, 2023-12-22:
I just tried your gcode on my N4Pro and the thumbnail showed up perfectly fine. I am almost sure, that it's the firmware. If you wanna be 100% sure, I am appending a sliced benchy with thumbnail, that I just generated on my Windows PC for you to test. If it also won't display, that will most probable be a printer firmware problem (maybe I will update soon to confirm this).

[EN4PRO_benchy.gcode.txt](https://github.com/Molodos/ElegooNeptuneThumbnails/files/13757336/EN4PRO_benchy.gcode.txt)

ruiaraujo012, 2023-12-23:
Hum ok, it's a shame that the firmware cause that but it's ok.

I'll try your gcode, thanks for sending it.

If it is firmware related, could you do a fix or do I need to wait for a elegoo firmware fix?

ruiaraujo012, 2023-12-23:
I can confirm, your g-code doesn't work (thumbnail), it is firmware related for sure.

Molodos, 2023-12-23:
Okay, that's bad. In that case we will probably have to wait for a fix from Elegoo. I will update my firmware soon to test it (and then use a EMMC backup to downgrade again if it doesn't work). If I can confirm the same issues on my printer, I will try to contact Elegoo.
Meanwhile, I just go one more Idea. I removed some stuff from the gcode, that my plugin added and is not necessary for the thumbnail to display on the touch screen (the line saying, which plugin generated it and the thumbnails for the web interface). Maybe there is some weird interference in the latest firmware. I am attaching another gcode for you to try out :)
[EN4PRO_benchy_v2.gcode.txt](https://github.com/Molodos/ElegooNeptuneThumbnails/files/13758904/EN4PRO_benchy_v2.gcode.txt)

ruiaraujo012, 2023-12-23:
I'll try again that gcode and post the results soon.

ruiaraujo012, 2023-12-23:
The same happens, it's most likely firmware related.

Molodos, 2023-12-23:
That's unfortunate. I will do some more research soon and then notify Elegoo.

PS: Just released the updated PrusaSlicer version :D

ruiaraujo012, 2023-12-23:
Thank you :)

Nice, I'll try the prusa version, should have the same problem, but I'll try.

Molodos, 2023-12-23:
Yes, probable the same problems with the firmware, but just wanted to let you know as you asked for it before :D

peterlvll, 2023-12-24:
I cannot seem to get the plug in to install.  When in marketplace and under the installed tab, the plug in is unchecked and is grayed colored.  If I try to check it, after clicking quit cura, the program crashes. 

Does my Npt4pro need to be connected to my computer or something for the plug in just install on Cura. 5.4 version BTW.

Any help tx

Molodos, 2023-12-24:
You do not need to connect the n4pro. What do you mean by "check it"? Is it disabled and you are clicking the enable button? Maybe you could describe it ab bit mor in detail and also provide a Cura log if it still crashes. Thank you :)

Molodos, 2024-01-10:
Closing this issue for now, as the fix patch for MacOS just finally also launched in the Cura Marketplace.

closed 2024-01-10 by Molodos