Molodos/ElegooNeptuneThumbnails #27: OrcaSlicer plugin

martin-badin, 2023-10-27:
**Question:**
Is it possible to create this plugin for OrcaSlicer? I would like to use this plugin but currently, I can't find any way how to generate the picture from OrcaSlicer. Thanks advance

Molodos, 2023-10-27:
Currently, I do not have the time to do something like this, but feel free to do it if you want to try it. Also after some research, for me, it looks line Orca Slicer doesn't even support plugins.

closed 2023-10-27 by Molodos

Wal-demar, 2024-10-25:
I have same Question:
But Orca have a Post-processing Script section.  I found a script from fifonic [ElegooNeptuneThumbnailPrusaMod for PrusaSlicer/OrcaSlicer](https://github.com/fifonik/ElegooNeptuneThumbnailPrusaMod)
but 1. it doesn't work on my Mac, and 2. it doesn't work on my Artillery Sidewinder X3 Pro.
Is it possible to run the plugin like a script after the G-code generation?
Thanks advance!

Molodos, 2024-10-26:
There is already a version of this plugin being able to run as a script for PrusaSlicer/Orca, that I made: https://github.com/Molodos/ElegooNeptuneThumbnails-Prusa
Hope that this is what you are looking for :D