Molodos/ElegooNeptuneThumbnails #28: Thumbnails in Cura 5.5.0 for Elegoo Neptune 4 Plus und Max?

prisma7411, 2023-11-06:
**Question:**
Model Elegoo Neptune 4 Plus ?

**Additional information:**
Add additional information (optional)

Molodos, 2023-11-06:
I definitely plan adding support for those printers if I can somehow get access to them. I am currently already in contact with Elegoo regarding that :)

prisma7411, 2023-11-07:
Thanks for your Info. Thats great

Molodos, 2023-11-21:
Quick update: Elegoo sadly did not want to cooperate with me to add the new printers. But maybe you could help me. I am pretty sure, that the thumbnail format is the same for the Neptune 4, 4 Pro, Plus and Max. Therefore you could try selecting the Neptune 4 or 4 Pro in my plugin and check if it will work with your printer (a photo of the printer screen would be really nice). If it works, I can officially add support for the new printers.
Thank you :)

AshTemp, 2023-11-27:
I am able to confirm that when sending a file over moonraker to the machine the thumbnail does show in in the preview in the Neptune 4 Max. I've got a long print on the machine right now but will hopefully remember to report back when I start the new print. What do you need to be able to add the new printers exactly, besides the printer screen? 
[image]

Molodos, 2023-12-05:
New printers are added, closing the issue :)

closed 2023-12-05 by Molodos