Molodos/ElegooNeptuneThumbnails #31: Thumbnails for Artillery X3 Pro?

manuuu332, 2023-12-21:
So, it seems the new Artillery printers use the exact same format that you are using for this though they have 3 different types of thumbnails - simage, gimage and also mimage

I image just the sizing of the thumbnails is different but the gcode looks like the same format string, except with the added mimage

Just wondering if this is something you would be willing to look into! No worries if not though!

Molodos, 2023-12-21:
Hi! Generally, I do not have too much time at the moment, but if it is really that similar, there might be a chance. The only thing is, that I do not own an Artillery printer and therefore would not be able to test it.

Maybe you could provide me with some more information so I can estimate everything a bit better:
- Which exact Artillery printer model do you mean
- You are talking about the same format. How do you know that? (I guess you just looked at the gcode. In that case I am interested in how you are generating those thumbnails until now)

I think that info could help me assess the situation a bit better :D

manuuu332, 2023-12-21:
Thanks for the quick reply!

Yeah so the new Artillery Printers (Sidewinder X3 Pro in my case) come with their own Artillery Cura 4.8.0 slicer which will automatically add the thumbnails in a certain format. After trying different thumbnail plugins I found yours which provided very similar results so I figured I'd ask you here, hoping it's something simple you may be able to figure out :)

I'm not versed in Python so I'm unable to figure out how your plugin works but comparing the two GCODEs (samples below) I can see many similarities between the two. Maybe you are able to make sense of this! Again, if it will take too much time then I'm not concerned about it as it's just a thumbnail, after all!

Of course, I could just use the provided Artillery Cura slicer but I'd rather use updated Cura as opposed to the more outdated version.

[ArtilleryCura_3DBenchy.txt](https://github.com/Molodos/ElegooNeptuneThumbnails/files/13734126/ArtilleryCura_3DBenchy.txt)
[Molodos_3DBenchy.txt](https://github.com/Molodos/ElegooNeptuneThumbnails/files/13734127/Molodos_3DBenchy.txt)

Molodos, 2023-12-21:
Thank you for that further information, exactly what I needed.
I must say, that this is really interesting, because Elegoo also has its own Elegoo Cura for generating Thumbnails which is also based on Cura 4.8.0. Also I had a look on some Artillery Sidewinder models, which are also pretty similar to the Elegoo models. Therefore I think there is a good chance that this could rwally work.
By the way: The outdated Cura version and wanting to use the latest features also made me develop this plugin:)
I will have a look into everything soon, when I find some time.

manuuu332, 2023-12-21:
Even their websites are suspiciously similar, hah!

Sounds good! Thank you for taking the time to look into it and I hope you enjoy the holidays! :) 

Let me know if I can provide any more information!

Molodos, 2023-12-21:
Hi, I indeed have another question. I cannot find any download for the Artillery Cura and have read, that it comes with the USB stick of the printer (interestingly, that's the same for the Elegoo printers :D). Do you know where to download it and it not, could you somehow provide it to me? :D
Thank you

manuuu332, 2023-12-21:
I don't believe it's available for download anywhere, but here I uploaded it to google drive - It is a 150mb .exe by the way!

Let me know if you'd rather me upload it a different way!

https://drive.google.com/file/d/1-8IovFewuiVs7Rm6FvY9Tbuq8XBFwo1_/view?usp=sharing

Molodos, 2023-12-21:
Google drive is perfectly fine, I downloaded it, thank you :)
I will let you know once I analyzed it a bit deeper.

Molodos, 2023-12-22:
Quick update: I just dug very deep into the Artillery Cura and its plugins to find out, that the underlying image encoding algorithms are exactly the same as the ones for the Elegoo printers, that's very good news. An it gets even better: The Elegoo version, my plugin is based on, used compiled binaries which I wanted to get rid of because of multiple bugs caused by them. The Artillery version already got rid of them, so I can even use this new research in order to improve my plugin. Win-win!

manuuu332, 2023-12-22:
That is awesome news! I'm happy you were able to make improvements to your plugin from this, if nothing else!

I will definitely be recommending this plugin to everyone I can! Superb job!

manuuu332, 2023-12-22:
Interestingly enough, it seems Artillery did actually release the files of the SD cards as well as released their own Artillery PrusaSlicer (3.0.1?) which also gives some information as to the thumbnail sizes as well, (85x85, 170x170, 230x230)

Doesn't seem they updated their Cura Slicer or even have it available for download but instead moved to Prusa, for whatever reason!

It would appear the encoding format is the same as the Artillery Cura Slicer at least based on the produced GCODE ..

https://www.artillery3d.com/pages/d

Molodos, 2023-12-22:
That PrusaSlicer thing sounds interesting, I will definitely have a look into it. Interestingly the Artillery Cura contains profiles for the X4 Pro and X4 Plus, which as far as I know are not released, yet. So it seems to be up to date at some extent.

Molodos, 2024-01-04:
Hi, I have good news for you. As the changes I made did not cause any new errors, I have decided to start adding support for the Artillery printers. Today, I can send you the first beta version supporting the X3 Pro. Maybe you could test it out if it works so I can start the next steps. Images are very welcome :)
The package, that you need to drag&drop into Cura is inside of the zip: 
[ElegooNeptuneThumbnails.zip](https://github.com/Molodos/ElegooNeptuneThumbnails/files/13836045/ElegooNeptuneThumbnails.zip)

manuuu332, 2024-01-04:
Seems to be working quite well! Nicely done.

I uploaded pictures of all 3 thumbnails so you can see what yours looks like as well as what the OG Artillery thumbnails look like!

[image]
[image]
[image]

[image]
[image]
[image]

Molodos, 2024-01-05:
Nice, thanks for testing so quick and happy to hear that! I guess the next steps are finding the right gray for the background to nigely integrate with teh UI as it looks like the same gray surrounding the image in all the cases. Now as I know it will work, I have also contacted Artillery if they are interested in a cooperation, so lets wait for that and in case they don't wanna help, we need to find a plan for finding the right gray tone.

manuuu332, 2024-01-05:
I think you're going to have a bad time trying to get answers from them as their customer service is nearly nonexistent..

However, they just used black for their thumbnail background and I think that would be just fine.. I don't think there's any need to match to the UI Background but that's just my opinion, it's your plugin after all :p

We could always do it the hard way which is you sending me different files with different shades of gray until we match it 🤣 but that might be way too much work haha

Molodos, 2024-01-05:
I think I will give Artillery a few days. If the don't respond, we can do one of your suggestions :)

mrlinux1, 2024-01-07:
I also tested your plug from above and it works perfectly.
Can you explain me what kind of file format is used for these images?
Many, many thanks for your work!
Kind regards, Thomas

Molodos, 2024-01-07:
> I also tested your plug from above and it works perfectly. Can you explain me what kind of file format is used for these images? Many, many thanks for your work! Kind regards, Thomas

It's not a common format but a format only used for those thumbnails. Why are you asking?

Print-Fuchs, 2025-01-10:
Hello Molodos,
at first thanks for all work and sharing it with us.
I tryed to change the original Cura Thumbnail .py script since days, but the printer only shows black pictures or the Artillery Logo.
So i was super happy to find your post and your stuff.

So i installed the stuff and my g.codes start right this now, but the printer still shows nothing -.-

; thumbnail begin 32 32 1796
; iVBORw0KGgoAAAANSUhEUgAAACAAAAAgCAYAAABzenr0AAAACXBIWXMAAA7EAAAOxAGVKw4bAAAE9E
; lEQVRYheWWa2wUVRTH/+fcfT+63d22rN0FJC0UYnBDCARYIX5QYojG1ASjJgohIUajIZBATJRXYkKi

Any Solutions for Artillery X3 + with Cura 5.9.0?

Thanks a lot!

Print-Fuchs, 2025-01-10:
Hello Molodos,

at first thanks for all work and sharing it with us.
I tryed to change the original Cura Thumbnail .py script since days, but the printer only shows black pictures or the Artillery Logo.
So i was super happy to find your post and your stuff.

So i installed the stuff and my g.codes start right this now and the printer still shows nothing -.-

; thumbnail begin 32 32 1796
; iVBORw0KGgoAAAANSUhEUgAAACAAAAAgCAYAAABzenr0AAAACXBIWXMAAA7EAAAOxAGVKw4bAAAE9E
; lEQVRYheWWa2wUVRTH/+fcfT+63d22rN0FJC0UYnBDCARYIX5QYojG1ASjJgohIUajIZBATJRXYkKi

Any Solutions for Artillery X3 + with Cura 5.9.0?

I Changed to the Sidewinder X3 pro and everything, but its still start with 32x32 and 200x200.
Thinks its need size 85x85, 170x170 and 230x230, but how i can chage this at your addon?

Thanks a lot!

PapadAlx, 2025-06-02:
I verified that it also works as a charm for Artillery X3 Plus.
Thank you very much!