Molodos/ElegooNeptuneThumbnails #32: Mesh tools breaks thumbnails?

coddiewomple, 2023-12-21:
**General information (please complete the following information):**
 - Printer: [ELEGOO Neptune 3 Plus]
 - Printer Firmeware Version: [1.2.5.1C & UI  V1.4.3]
 - Cura Version: [5.6.0]
 - Plugin Version: [3.6.1 & 3.6.2]
 - Operating System: [Windows 10]

**Describe the bug:**
Load a STL file into cura and slice it, the thumbnail is displayed ok on the lcd. Load the STL file and use mesh tools [3.7.2] to split into parts and thumbnails will not display on the LCD.

**Steps to reproduce:**
1. Load STL: https://www.thingiverse.com/thing:6269202
2. Mesh tools: Split into parts for per model flow settings
3. Save: Slice and save to SD card
4. No thumbnail shown on LCD

**Additional information:**
Have been having issues with Elegoo Neptune 3 plus for months and have recieved new parts from Elegoo to fix variable Z offset issue. At the stage to calibrate E steps and flow for good quality prints, printer is running latest beta FW and slicing using latest Cura version 5.6.0. Today i find when using mesh tools [3.7.2] to split the model into its parts for adjusting flow rate per model for calibration testing the thumbnail does not show on the lcd of the machine however the preview in the cura extension window works ok. I noticed today thumbnail update so removed 3.6.1 and upgraded to 3.6.2 but still the same problem remains when mesh tools is used to split into parts. I tried all cominations of merging and regrouping the models back togethor but not thumbnail on lcd.
 I assume there is some complexity going on when mesh tools is used to split a stl into its parts. Im sure mesh tools did not cause this issue under cura 5.4.0 with the thumbnail Version that was installed with that, i think it was 3.4.0 but that doesnt work with latest cura version.

Below are the sections of gcode from the STL sliced before using mesh tools that works and after that does not show thumbnail on lcd provided for anaylsis.

**Sliced and working thumbnail (mesh tools not used)**

```
;gimage:0`000<P000380000?<?M1CH1001Z2P00000000000039<DBCFLhTRnXa:c[9:FBCI9_:<DB;59dAO1S6Th`haW:<Y:<eY@Xj24;DU1FUi;?UN~B[jCWg_LeBekg@LbQ:MZgdW2iK1H>4VnGTi8:4X`B;9;c5LPiK]kEEYJTY[4X4]0GUiMc4NTMBi7ZUe<GLYVYBQ4]2Kf=~@YJe[E:Te<JSI<B4JRMkILA4a0Bl18?CU?RmS4Y[@[JekEZ]BPM2YY~7BRG]I<b5c2F;j3VUg2IB:4;6L^TaAUX6[8IRa=aWFRU2Kf_G]C:4DXcaNhPYLhb7a<QZi>CXJ^M1Y6Y9BRF32D88LnJS9E89>PG]Qi<5]6IJiZ^4c6UBY787LbM:U5]SJTQ:J5;We8CDYV8Ta0Pj99=UHZEbi=cdJlS<R5Z4LXMREJg]DTiST6^?JmBLlG>cU9J]1L=C941U1<<Y82Dc:B06`20Y@2DYhBkP0<DT:@359P3o0?l0o`3o00URhBgP00dR`>4Qh6:0:b80Gh803SaRhBcP02m2;^4mh4802blRX4;T>^<Vh49P:b80GD;R;2_Q?2cP@P0;?48~?D806S<@2P08hBgP8X0R00YRP2TR:F12@2~R05`R=~3Q;N0R00XRhbGPP2;Q>n006BT22P08@X1200OQ:n1282:0;2:0@T0[8T3R:^0Y05L]8T22@2;R:N0022TRX4:0<bD05B8^P4802B;Q?>20@T0b8V3Q:ccP@2802R8]@2_205HR@2:08RTP8P0?8^8Xh43S9^22hbCP8P0B928Y@3~R00XRX3HR@2cT>N001b801n4ji3SP@V0[@V1205DR828_P>4Ph28P8P0>;B9P8^4mh61201@Ri3OP@P0;8Z0`8P0;8P07PRU0:b:0@P1Ehb?P82;08R0m00gS9>0R:d12P>4ih28052A200`RX3DR00~R00Y2@2~RP2lR05DR82;Q>^2P8R0R00cR;>0RhB3PH4:P8P0DH^4;gimage:Qh00;8RB0@P0;8P0;8^8Xh20[8X0_8P1E8R3T=^0RP2;Q;N0P8P0;@^@eh812X2804c~R82@R900:8S20@P0;8X3R:n4Ph:0R=R0[8X0_8P1E8^4[h212hR[Q>^128>@dh2802T8UX4:092804T:08^4Qh009;b:08RD02b:0;R:0?290:b:0;b80ER9P:c9P8P0:@P07hB_P8X12014k8Z0T8R@02D90@V0b8P078Z12@>@ch490:b:0;b80ERTRhB7PP3HR:@09928bcT12hC_P82m2013S9N022P09:H:0<R801b;0P^4Qh60[8X0_8P1HhB3PPS802`89H392hBcP9B80a^<Rh280`>4ih2:0@P31HRm2<`32<n@bhB`[h03o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`3o0<CS8N1S0;ld0`WQ>>00^><Ph0<@0;@h0aCQ:^00]0<E0:DdPc@02N8Wh0<@9f<o09gT<N4Zh0<:>00:0`_T<>13=n8Vh6<ghBWP09SQ:^034@0:0`XaHn8UhCOT;n00VN4Xh0<DhROP00[38N8ohCKP0:3Q=N035@0:8n4Wi2h]h02EhSkPXc@02N4dh0<>hC?P@n4Xh6<j09gQ:^032SP02P<>hBOT;>0XHn4Yh02G0a<02^4eh:?Q:>338N4b=^0054<0Pn4Xh0XE00[T:n0:1c7Q<^8mh00C?`<7hCCP08;R?>015N4hh00:8n4YhR@Sh00B=>8nh04>07<hP@0:i2[P0@kQ9^118f7Q>>006cSR9n014cP0JcX12n8kh00:0@TS@N4ch47Q<N8oh64WhC3P01_T:N015@1U?n<Ph04@>P0:8`48>B4iHCWR9CgP01<o8`0:8`4@<CUQi2SP063R8^015N8jh00:hBKP0@ST9n4UhR?P01?T9^011`;gimage:0:hCCP0@~SHCTQhBCPP@1PhBKP0AGQ:^002^4ch67T9B@Sh00C>04<00[R>n014>4Wh67R8N4fh01A>^8ih6402^8hh04>:04701dj0A@02R<132SR=bCP05;T8^0122<02^8Rh048=d7T8N012P0KhBSP0AH02^@Ph047>N4ghR3P057Q>>013^4Sh00:hSSP`C4g<B=1hBWR9^1QhRKQ<N8Sh00MhcoP0AHj00XShC0ghckP054h8`4ChB?P00[S?N011n4`h2MQhClahccP01CQ8n1Q00Xd0@oQ9^31>01N0AL02j7Q934fh00Dhc_P0@[R>N002R<14>4Th84W05gR=^015^8eh00:hScP=n8dhc[P01<o0A0o00[Q=N21@f4i8CN1hBCP>N8Ph01Q0@gQ<n012>4Sh00L8`4EhB?P00[S>N31=f4WhS<Ph01W0@Pa0@gS>>006cX75^8eh00;an4ohcOP06c11`WQ9n<fhSCP01CS=N0W00~Q1aH02n8gh01bhcCP1`XghS;S<n005>4dh<L02b473S7Q9n3707kR=^27hS<bhc;P01CR>^07300;hBKP1`LQ9b472n8Uh01nhR7Q=n<ah00Dhc3P1a402n4eh<MA9b4W:8OQ?c;S;n00T`LG00_S;^082C7Q<^8Th02IhS7P21Lj00[R8^28=n4UhS3P09kS;N085`0;=><~h02T=0PG0;7R;n083n4Th:PahBGP0;7R<N082R>XhC0UhS3P0;Dd20[Q:N<[:^00^cBXhR7Q9N<Qh030hRoS:N00o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`0OhRkP@P0<?B8_:@1>`Rh02N8^h48l00~U@RD0@d8^hCgP@P08;D8~HP1=aC<0248bhCgP@P09@^4i8N12047Q>n0RH;;gimage:><Rh280228_H>4jh28c04lY00[Q>n0RP28l00OR:N0RH>4_h280;RT22P08@X12`2TRX4:PhCkP:F0Y?008hRgP:F0YhR[P03=29009@X1200LR983Q8^30hChlh80l00SQ?^0YH2Tl01<Y0PP02B;R;N20@P0=8Se08RF2888n@2~R<X8PPP0a?V8T00Q2P4;092802B8UPR22?V0R9H8PPSh04RC200XRX3HR00`]8V12H4;Q;n10@T0[@V12H480;l8T00LR<X0f8~0e8R22hR_PH49P@^4_h43R:N0RH49P@^4_h41201<TPP0;8Z0^8P0;hCWP8RaP8^4Rh612H3DR@2~RP49P=B80<R802R;S:>20;R;0>d8m@49P8^4Rh612P29P8^4Rh612H3DR01@T@P0<8Z0e8P0:hRcP8^4Ph80RP28_H2`R@2~RP28_H2`R038R00XR=H0e8~3Q;>0R?X0R;D0RP4:08V0RP28_H2`R01@k@R@02b:P?B802D8~X2:08ReP;290:b:08ReP;280<R802R;S8n20?B;0=B:P@T0RP4:08V0RP28]H2`R01<kPR@02R8`P480248U`2:08ReP;290:b:08ReP;280<R802T:0@P078Z12@2:0@X0RH2:08ReP;2804S_29009@X0R?P07@P088X0R;F0~8T0[8X0R;F0~8P0b8P0:@X1200LRhRSPP28[@2:0@X0RH2:08ReP;2804C~222@02D90@V0f8T0T8S;>@2:08ReP;290:b:08ReP;280<R802d90@V0f8V3Q8^0RhBgP@49P8X12P29P8X0R;F0~8P0@hB3P0PX02CF2:F0f8T022D0RP28]H2`R@2~RP28]H2`R038R00~TPRUP=R:0;H8cH2:0@X0RH2:08ReP;280o`3o064V:@36hBkP92TV0<?QC^0c92H0`n4^hbL;0000000000000000000000000000000000000000000000;simage:0`000:00002P0000?<?M1Oh0003`1P00000000000039<DBCFLhTRb~jI9=TVlXajS7=DQ5l;V>TXnTi18>cU4]2><JCS>GT1;@T_0iKKf<5iHBKA8~8@^Ta:4YBQ>eBbBT:>RG]Qi<5PlB[1;b8FUJ]S4Zg]C64m9a~@]>DJ5:Te>GL9H^4LTN3QV84RhBS9g]4a8C<2C[6XfC4YMBVJ^B2aMc5LTMB23ZF]BiK5JF`Jk1ckEY_JhPYZBWh_O5kLhb?JfC<CCZTZlGDY7;6L^PiQFZ7F^MbQLb4e0FdI69Ua8=ZBD[T~`Jd2D8TPbU2YV:UJVMJ1ZcTNPMcKCY8BZJKIV:XH~CLe9@eYABM[DZF[JeBCf=[@]NmTi@HaPBLhD7P:214:02J9bPP9>4Ph02N:02NhB3P0?l0o`2IHP0;8X0o@^<nh60R04cS?N12<P0;;d;Q:^009B;S?>0P9b802D9PhR_P8T0a8T0R04[R:^0RP28o00Q2@4805@8800LRH4;0@RTRH2:08T0R04XR82Q2hc_P828022:08P0E9b:08SK09b9P:B908V0R82TR00PR@28XH^8Yh0148R0W8T0R920R00`R@>8Yh68W8P0?8T0R=P07?R:08P08:B:PhB7P@V12H280A28P8X0R82802d908V1200lX@SH022lRP28022TRX2oS>^12@2:08P148R0RP28P8P0:@T0_8V3S>N0R011200WR:n0RP28022TR00PR:B0RP280A28P8X0R82802D9P8^<hh80R00lR@2802B9P<R8022TRH>8Xh80RhROP82:08P148T0RhBWP<2908P08@X3Q:^0RH2PR00hRhRKP@2PR00PRH28a00PY8V0RhcOPH2908X0R04DR@><f=N108P08@~0RH4803@8800OS=>0RhB[P@V0RX2TRP4;R9N12@2:08P16@T1200W2@35R?b807RL04bL0DTL05clR09XRH2;Q:>00VZ80o`3;simage:o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`2ihRCP``2D?@<<;@2@0a40Qe`01bh3325S087R8n031`080`LeHb4S9V<0N^8Rh0<=00P322aShR7Q9n00N^8Ph0<@00SQ?n2Shc?Q?^00M^<bh3Dk00OS<N033N4Vh6<02><`h01f;@<800P32b4S;6<0NP<>00P32B53hCgR8N0l00hk8`1Y0A4]00OR9>2Q=>4lh00?hboPVV<0H>4kh007;@4A00SS;^0chbgP00oQ?n013>8Sh01I`B<0204>8d4S01DS0A40DP4>00SS;>011cWQ>^1QhCWS:n003cd02><Zh04=;>4Uh47R8^00C@4B00SS:N31=3?Q>>003n<Xh:4m00P13>4Uh24SH@1=0A8023cQ=n0hhbOP00oS9^0130088f4^8h4iHBKQ?^<Uh00o?J4S00P^0@~^XBd05N4fh04A>`080@L~hbCP047Q=N012^4Th008;Z4iHCCQ>N21>>4dh00FhCKP0A7Q=N00247Q9SSP040S0A0S00SQ<n2QhCOP@N<Sh3SS8^003bf100T14P1;hb7P0A8022iQ>><Ph00@hC;P0@T02@478h4~hCgPHC@0Bn4ch04;=J7Q9>00604@00TS0@[Q9SCP053R?n31;04:hSkP01HQ1A7Q<^0022j5hSgQ8n00ER>1IBb5=N8lh00@hC_P;@088@DA8@08hS_Q?>00F`D7hC4Sh00@hS[PaBd02>8ih0D9hR3P9CG506DShB8Wh00@hBCP1@d02>8hh:GQD>0QABaUhCX_h01e1A;R=n002>8fh0D8>N8ehBOP07X64`09A^8dh3?Q8n00P0HB9@08>`251POQ8^16hB;P1PP0SPH==3?R<n00T@H8hS;P<n4_h02FA^4U<N00o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`3o0?l0o`;;simage:3o0?l0o`3o0?l0o`3o0?l0o`3o0?]R9`07;f;R:^00?Z801c9R00PnHP0e8^4Yh40R`>4Yh29P8P199b90<B;0@T3Q:>0R02D220078V12X2;R:>1P8S40?b8022;R<N1P8~0RP2809~801c0RH30R00XRhBkP828XH^8`h8908RQR88809f801b:08S2P8P07X^4]h68Y@:;Q;N1R0122=P07?R:08P0:8T12@2@RH290@T0a8V0R02IG8UO08X0RhRoPX:9P8^4Xh412@4908RQ0@T12011200Tb8X0R00PX8RA08RmP8V0R@28_@2@RH280:28022:08S:P8^4~h40X8T0RH28`@>4[h2908V0R<43 [... cut ...] 000000000000000000000000000000000000000000000000000000000000000000000000;Thumbnail generated by the ElegooNeptuneThumbnails plugin (https://github.com/Molodos/ElegooNeptuneThumbnails);FLAVOR:Marlin
;TIME:799
;Filament used: 0.659279m
;Layer height: 0.2
;MINX:91.587
;MINY:136.677
;MINZ:0.2
;MAXX:238.413
;MAXY:193.323
;MAXZ:0.6
;TARGET_MACHINE.NAME:ELEGOO NEPTUNE 3 Plus
;Generated with Cura_SteamEngine 5.6.0
```

---------------------------------------------------------------

Thanks and apologies

Molodos, 2023-12-21:
Hi!
I just tried replicating the exact steps you mentioned, but for me, everything worked just fine. Maybe you could share the whole gcodes instead of just parts with me, so I have the whole gcode for testing.
Thank you :D

coddiewomple, 2023-12-21:
My apologies,  here are the two complete gcode files.

 Just to highlight my N3 plus is on the very latest beta fw with latest UI to my knowledge which Elegoo pointed me to to help solve Z offset variance issues. If these Gcode files both display the thumbnail on your test machine please advise what fw you tested on, im happy to test them on my machine under same fw in reply. Thanks.

[coddiewomple gcode.zip](https://github.com/Molodos/ElegooNeptuneThumbnails/files/13744103/coddiewomple.gcode.zip)

Molodos, 2023-12-21:
No problem, you could not know :D

I have tested you gcode and the thumbnail showed fine on my machine. But I use a Neptune 4 Pro (it uses exactly the same image encoding format as the Neptune 3 Plus, so that shouldn't be a problem) for testing, so my firmware will probably not help you. If you still want to know, my firmware version is 1.1.2.03 and UI 1.2.4.
I think that the errors you are experiencing could very likely be because of the firmware you are using. I guess at least it can hardly be the slicer or my plugin if the same gcode file works fine on another printer. Therefore you might wanna wait for the next update.

Best regards

Molodos, 2023-12-23:
If you wanna know for sure if it is a printer firmware related issue, you can try Elegoo Cura. If that one also does not work, it is because of the firmware. I am happy to hear about your findings :D

coddiewomple, 2023-12-23:
Hi, thats a sound idea but i just downgraded to FW 1.2.5.1B & UI V1.4.2 and i am pleased to report the thumbnails work fine now so apologies for not testing it before but this machine is now causing much aggrrevation.

I found the C version of beta FW was allowing thumbnails sometimes but the info on them was all wrong so decided to re install cura 5.6.0 wondering if it had got messed up but only after purging Cura completely from the machine by manualy deleting what was left in the localdata folders after uninstalling. Trouble is now i dont have reliable way to browse directorys from the printer on SD card as the multifile feature doesnt work on the downgraded UI version required for the downgraded FW. My head spins.

Thanks for the plugin btw.

Molodos, 2023-12-23:
No problem, you could not know. Closing this issue for now :D
And good luck with getting everything to work again.

closed 2023-12-23 by Molodos