Molodos/ElegooNeptuneThumbnails #35: No Thumbnails on printer screen with latest Elegoo Firmware

hallard, 2024-01-24:
**General information (please complete the following information):**
 - Printer: Neptune 4 Pro
 - Printer Firmware Version: 1.1.2.53 UI 1.12.11
 - Cura Version: 5.5
 - Plugin Version: 4.0
 - Operating System: Mac OS Ventura 13.6.1 (Mac Mini M1)

**Describe the bug:**
As described in #24 with latest firmware, all is fine (thumbnails are generated and ok on moonraker fluid interface) but not displayed on printer screen

Molodos, 2024-01-24:
I think this might not even be a bug with my plugin, but more likely a bug wieh the printer firmware. I have heard befire, that there seems to be problems with certain printer versions.
Might I ask you to check out if thumbnails are displayed when slicing with Elegoo Cura?
If so, it must be a bug with my plugin. If they are also not displayed, the bug is inside of the printer Firmware version unfortunately.

hallard, 2024-01-24:
Sure, your Thumbnails works fine on generation and correctly displayed on Fluid interface 

[image]

[image]

> Might I ask you to check out if thumbnails are displayed when slicing with Elegoo Cura?

Will check that one, you mean Cura Version 4 so? I'll send the official benchy gcode from elegoo usb key to see also

Don't know if it is a printer FW bug or a printer FW format change, but surely I'm pretty sure it's not on your side.

Molodos, 2024-01-24:
Yes, I mean the Cura version 4 thing. But the pre-sliced benchy should be alright as well.