Molodos/ElegooNeptuneThumbnails #39: Elegoo Neptune Thumbnails 4.2.0 with Cura 5.6 icon shows only small parts not the whole object

rmayergfx, 2024-03-14:
Cura 5.6 together with OctoPrint in use. Sometimes the Preview Thumbnails looks totally strange,
just a part  of the object and not the overview. The only help at this point was to restart Cura again
and then the Preview looks like expected.

At this point, i am interested how the Thumbs are created. Sometimes the angel and viewpoint is not the best.
is there any option to change the created thumb?

rmayergfx, 2024-03-14:
Strange that the viewiing angle makes no differenz, so i don´t know how cura is making the thumbnails. Looks for me that cura is always using the isometric view, equal how the parts are previewed or placed. I never was able to change the thumbs.

Molodos, 2024-03-14:
That's weird indeed. But as long as the parameters around the thumbnail, that my plugin adds, look fine, I guess it's some kind of Cura bug