Molodos/ElegooNeptuneThumbnails #44: Cura 5.8 Crashing when change parameters and thumbnails don't work

grandpurs45, 2024-08-18:
**General information :** 

 - Printer: ELEGOO Neptune 3 Pro
 - Printer Firmeware Version: 1.1.5.1b
 - Cura Version: 5.8.0
 - Plugin Version: 4.4.0
 - Operating System: Windows 11

**Describe the bug:**

Sorry for my english i'm french :)

thumbnails don't work since I update Cura in 5.8.0.
if i change a parameter (check or uncheck a case) cura crash

**Additional information:**

[image]

event viewer windows : 

2 events : 

first : 

Nom de l’application défaillante CuraEngine.exe, version : 1.0.0.0, horodatage : 0x66ab5087
Nom du module défaillant : ucrtbase.dll, version : 10.0.22621.3593, horodatage : 0x10c46e71
Code d’exception : 0xc0000409
Décalage d’erreur : 0x000000000007f6fe
ID du processus défaillant : 0x0x8480
Heure de début de l’application défaillante : 0x0x1DAF153CDD0A11A
Chemin d’accès de l’application défaillante : C:\Program Files\UltiMaker Cura 5.8.0\CuraEngine.exe
Chemin d’accès du module défaillant: C:\WINDOWS\System32\ucrtbase.dll
ID de rapport : 2068b109-b089-46c6-959c-eac9e4ee6850
Nom complet du package défaillant : 
ID de l’application relative au package défaillant : 

second : 

Détecteur d'erreurs 1924136390620708749, type 5
Nom d’événement : BEX64
Réponse : Non disponible
ID de CAB : 0

Signature du problème : 
P1 : CuraEngine.exe
P2 : 1.0.0.0
P3 : 66ab5087
P4 : ucrtbase.dll
P5 : 10.0.22621.3593
P6 : 10c46e71
P7 : 000000000007f6fe
P8 : c0000409
P9 : 0000000000000007
P10 : 

Fichiers joints :
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.20e5d9a7-d8dd-4c7c-ac19-1801edf33aa6.tmp.dmp
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.f4453f6d-f7bb-49ea-a5f6-b7235e724a4a.tmp.WERInternalMetadata.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.69fe2ee8-19c7-45d8-beb4-0701a82e3000.tmp.csv
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.1d89d519-e655-4310-8c3e-0ba3584c038c.tmp.txt
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.40756123-1a88-470b-8db9-8892c8b5bf16.tmp.xml

Ces fichiers sont peut-être disponibles ici :
\\?\C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_CuraEngine.exe_67121b3b28a69fadd13c15d508212c3de37b8ac_30f897e4_3dae0576-8526-4887-83e8-00c510e757a7

Symbole d’analyse : 
Nouvelle recherche de la solution : 0
ID de rapport : 2068b109-b089-46c6-959c-eac9e4ee6850
Statut du rapport : 268435456
Récipient avec hachage : 7a56a51467ed25c6fab3e7da4e597f8d
GUID CAB :0

I hope my information can help you

grandpurs45, 2024-08-18:
I uninstalled the old versions of cura that I had (5.6 and 5.7.2).

I also deleted the plugin to reinstall it and the preview works on the printer (but not on Cura).

however, as soon as I change a setting, cura always crashes.

[image]

subnetz, 2024-09-12:
I've got then same problem with
Printer: ELEGOO Neptune 3 Plus
Cura Version: 5.8.1